heuristicus/spot_ros #3: [Documentation] Deprecated Reference to Application Token

bd-bbeddard, 2020-09-21:
As of Spot v2.0.1, Spot no longer requires an application token. We now utilize a license that is local to Spot.

Would you be able to adjust the documentation accordingly? http://www.clearpathrobotics.com/assets/guides/melodic/spot-ros/ros_usage.html#get-an-applicaiton-token

Thanks!

dniewinski, 2020-09-22:
Just FYI, the source for these docs is here: https://github.com/clearpath-docs/spot-ros-sphinx-tutorials

I haven't had a chance to test the latest version with the ROS driver, but once I get a chance, I will make this change and any fixes in the driver

bd-bbeddard, 2020-09-22:
Thank you! Sorry for the wrong repo.

dniewinski, 2020-09-23:
All fixed up.  Minor change to the driver too

closed 2020-09-23 by dniewinski