heuristicus/spot_ros #5: Plans on adding the trajectory command

EricVoll, 2020-10-30:
Hello 👋
thank you for open sourcing this ros wrapper!

I'll be working with spot the next few months and will need to command trajectories. As far as I have seen, this ros wrapper does not yet support the `trajectory_command` as in this [example from Boston Dynamics](https://github.com/boston-dynamics/spot-sdk/blob/28f6d7f39145c2aaed127eeaa101ebd27b047182/python/examples/frame_trajectory_command/frame_trajectory_command.py#L80).
Are you planning on adding this feature? If not, are you interested in PRs, which add this functionality?

Best regards,
Eric

davesarmoury, 2020-10-30:
Hi @EricVoll .  The driver currently doesn't support this, but we are always open to PR :)

dniewinski, 2020-10-30:
@EricVoll the driver currently doesn't support trajectories and we don't have plans at this time to add support.  We will happily accept PRs that add this functionality :)

closed 2020-10-30 by dniewinski