heuristicus/spot_ros #9: spot_ros.py will shut down with exceptions when async tasks raises exceptions

sktometometo, 2020-11-12:
I got an error with spot_ros.py like below. Spot went down when the error showed up.

```
[INFO] [WallTime: 1605174374.227650] [node:/spot/spot_ros] [func:SpotROS.main]: Starting ROS driver for Spot
Traceback (most recent call last):
  File "/home/lshinjo/ros/catkin_ws/src/spot-ros/spot_driver/scripts/spot_ros.py", line 391, in <module>
    SR.main()
  File "/home/lshinjo/ros/catkin_ws/src/spot-ros/spot_driver/scripts/spot_ros.py", line 372, in main
    self.spot_wrapper.updateTasks()
  File "/home/lshinjo/ros/catkin_ws/src/spot-ros/spot_driver/scripts/spot_wrapper.py", line 346, in updateTasks
    self._async_tasks.update()
  File "/home/lshinjo/.local/lib/python3.6/site-packages/bosdyn/client/async_tasks.py", line 39, in update
    task.update()
  File "/home/lshinjo/.local/lib/python3.6/site-packages/bosdyn/client/async_tasks.py", line 96, in update
    self._future = self._start_query()
  File "/home/lshinjo/ros/catkin_ws/src/spot-ros/spot_driver/scripts/spot_wrapper.py", line 134, in _start_query
    response = self._client.robot_command_feedback(self._spot_wrapper._last_stand_command)
  File "/home/lshinjo/.local/lib/python3.6/site-packages/bosdyn/client/robot_command.py", line 281, in robot_command_feedback
    **kwargs)
  File "/home/lshinjo/.local/lib/python3.6/site-packages/bosdyn/client/common.py", line 225, in processor
    error_from_response=error_from_response, **kwargs)
  File "/home/lshinjo/.local/lib/python3.6/site-packages/bosdyn/client/common.py", line 340, in call
    return self.handle_response(response, error_from_response, value_from_response)
  File "/home/lshinjo/.local/lib/python3.6/site-packages/bosdyn/client/common.py", line 348, in handle_response
    raise exc
bosdyn.client.exceptions.InternalServerError: bosdyn.api.RobotCommandFeedbackResponse (InternalServerError): Command not found, id: 235
[INFO] [WallTime: 1605174455.007530] [node:/spot/spot_ros] [func:SpotROS.shutdown]: Shutting down ROS driver for Spot
[spot/spot_ros-4] process has died [pid 6430, exit code 1, cmd /home/lshinjo/ros/catkin_ws/src/spot-ros/spot_driver/scripts/spot_ros.py joint_states:=/joint_states tf:=/tf __name:=spot_ros __log:=/home/lshinjo/.ros/log/e6444882-24cb-11eb-a312-3868938abb0a/spot-spot_ros-4.log].
```

The exit of spot_ros.py is caused by exceptions from robot_command_feedback() function. But it seems to come from invalid value in the _last_stand_command member variable of SpotWrapper class.

sktometometo, 2020-11-12:
Is this because there is no exclusion control for accessing member variables of SpotWrapper class?

dniewinski, 2020-11-12:
"spot went down".  What do you mean?
Did the error show up right when you started the driver, or after you were already using it?
What version is the Spot software?

sktometometo, 2020-11-13:
I mean that motors of spot has shut down and spot has fallen to the ground.

The error showed up when I was controlling spot by teleoperation some minutes after launching the driver.

We use spot firmware v2.0.2 and python sdk v2.0.2.

sktometometo, 2020-11-13:
We can fix this issue with https://github.com/clearpathrobotics/spot_ros/pull/10.

dniewinski, 2020-11-16:
PR accepted.  Thanks for the addition :)

closed 2020-11-16 by dniewinski