heuristicus/spot_ros #19: Spot refuses to move after a while

grammers, 2021-02-01:
When using spot_driver to operate spot is it working great to start with. But after running the spot_driver for a while will spot refuse to move. It clearly receives movement commands, it attempts to move but stops mid-motion and return.

Have anyone had the same issue? Any idea of how to slow it?

Best regards!

dniewinski, 2021-02-03:
I haven't seen this issue with our spot.  Some questions:

1. How long is "a while"?  
2. Are you running any other controllers?  
3. Does it sit, or stay standing?  
4. Are you trying to control it from the terminal or from rviz?

grammers, 2021-02-04:
1) about 6 min

2) I have an interface that publishes cmd_vel commands, similar to using the terminal. No rather controller is active.

3) Spot remains standing, I can still make it sit and stand. But when I try to send cmd_vel commands is Spot not responding properly. It may raise a leg and put it back in place. The issue comes gradually, at first is it only in one direction it has problems to movie, but soon (1-2 min) won't it move in any direction.

4) See 2

dniewinski, 2021-02-05:
Hi @grammers .  I will be able to do some testing with our spot next week and can hopefully track this down

dniewinski, 2021-05-03:
Finally going to be looking into this. This seems to be an issue with the more recent BD software versions.

dniewinski, 2021-05-07:
@grammers please try the [latest version](https://github.com/clearpathrobotics/spot_ros/commit/731890bcc526af6c9befb556558c9ef59be6a2ec) and let me know if it fixes the issue.  It resolved the problems I was seeing on my end.

dniewinski, 2021-05-14:
@grammers Please try the latest version of the code.  I've been using extensively this week and haven't run into this issue

closed 2021-07-08 by civerachb-cpr