heuristicus/spot_ros #23: Robot cannot stand up but can claim it or power it on/off

zhaozhongch, 2021-03-04:
Hi Thanks for this work.
I had a problem let the robot stand up. Following the guidance from [tutorial](http://www.clearpathrobotics.com/assets/guides/melodic/spot-ros/ros_usage.html).
I can call rosservice to claim this robot in my terminal
```
~/spot/spot_ws$ rosservice call /spot/claim 
success: True
message: "Success"
~/spot/spot_ws$ rosservice call /spot/power_on 
success: True
message: "Success"
```
I can hear that the power is on from the Spot
But when I try to call stand up service, there is an error
```
~/spot/spot_ws$ rosservice call /spot/stand 
success: False
message: "bosdyn.api.RobotCommandResponse (UnsupportedError): The API supports this request,\
  \ but the system does not support this request."
```
Seems Do I need to upgrade something?

Thanks!

zhaozhongch, 2021-03-04:
My current solution is to use the previous version of the spot ros `tag 2.0.0` and then install the python package with the 2.0.0 version
```
pip3 install bosdyn-client==2.0.0 bosdyn-mission==2.0.0 bosdyn-api==2.0.0 bosdyn-core==2.0.0
```

Then control spot to stand up is successful
```
~/spot/spot_ws$ rosservice call /spot/stand
success: True
message: "Success"
```

I am still looking for the new version that can work. I use Ubutu 18.04 and ros melodic

sktometometo, 2021-03-05:
I think [current commit](https://github.com/clearpathrobotics/spot_ros/commit/0cfb0c1c43460238bc2ca113b1d567d6dd1c1985) breaks compatibility with base platform version<2.1.0 since synchro_stand_command() and synchro_velocity_command are available from 2.1.0.
You may have to revert the commit or checkout some commit before it or upgrade your base platform.

closed 2021-03-29 by zhaozhongch