heuristicus/spot_ros #28: Spot does not move with Interactive Marker

andu215, 2021-04-19:
Does anyone know what the error can be when the robot can not be controlled via the interactive markers? There are no problems controlling the robot with the controller or the console.

As soon as I pull on one of the arrows, the robot makes a step on the spot but does not move in the desired direction. (It seems to be blocked). Also, the depth image shows walls around the spot even though the robot has enough space.

Can anyone help me?

dniewinski, 2021-05-03:
Seems to be the same issue as #19 .  Hopefully resolving this week

dniewinski, 2021-05-07:
There have been improvements in the latest version (see #19) that help with movement issues.  It might help with your issue, but I suspect the issue is more related to Spot thinking it is blocked.  Please make sure:
- The cameras aren't blocked at all
- The space around spot is unobstructed
- The robot has enough light and features around it (not a blank white void)
It is odd that it works with the controller and the console, but not the interactive markers.  The markers are just using the console in the background.  Because the robot takes a step at all, means that ROS and networking are correct.  Try changing the Fixed Frame in RVIZ to a different frame (ie, body)

dniewinski, 2021-05-14:
@andu215 Please try the latest version of the code.  I've been using extensively this week and haven't run into this issue

closed 2021-07-08 by civerachb-cpr