heuristicus/spot_ros #33: cannot import name 'SpotWrapper' from 'spot_wrapper'

baldash, 2021-05-12:
I was following the setup tutorial and when running `roslaunch spot_driver driver.launch`

I'm getting this error:
`ImportError: cannot import name 'SpotWrapper' from 'spot_wrapper' (/home/baldash/Documents/ros/catkin_spot/devel/lib/spot_driver/spot_wrapper.py)`

I was trying to run spot ros using ros noetic, maybe is this the issue ?

dniewinski, 2021-05-12:
Hi @baldash . There were a bunch of changes related to this just yesterday.  Please make sure that you are on the latest version of the code.  Also, please delete the build and devel folders from your workspace and rebuild, just to make sure everything is linked properly.

baldash, 2021-05-17:
Thanks for your answer, was about to try that but it seems your tutorial on the spotros package is down ? I can't access it anymore

dniewinski, 2021-05-17:
Yeah, there was an issue with the build server.  I am rebuilding and republishing it as we speak.  Should be done shorly

baldash, 2021-05-18:
thanks a lot, gonna try to redo everything from scratch and see if I still have the error

baldash, 2021-05-18:
Ok for now running this on ros noetic is impossible so I don't think this issue is relevant anymore, gonna close this and assume that with the proper rebuild and a clean install of melodic the problem most probably doesn't exist.
Thanks a lot for the answers !

closed 2021-05-18 by baldash

dniewinski, 2021-05-18:
Hi @baldash .  When you say "running this on ros noetic is impossible", what do you mean?  I'm running it on Noetic without an issue.  What error are you getting?

baldash, 2021-05-18:
When running `rosdep install --from-paths src --ignore-src -y`, I get :  
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tf2_kdl: Cannot locate rosdep definition for [orocos_kdl]
tf2_sensor_msgs: Cannot locate rosdep definition for [python_orocos_kdl]
tf2_geometry_msgs: Cannot locate rosdep definition for [orocos_kdl]
```  
  
Which makes sense, because when checking the packages on the wiki, (for example http://wiki.ros.org/orocos_kdl), I can see that they are not available for noetic

dniewinski, 2021-05-18:
Please check the spot docs.  There is an updated section about building geometry_msgs from source

baldash, 2021-05-19:
Yes that's what I did, I cloned the spot_ros and geometry2 projects, switched to 0.6.5 on geometry2 and when running the rosdep install I'm getting the above error. 
I'm talking about the "Building the driver from source" section, are you talking about another one that I missed ?

dniewinski, 2021-05-19:
Sorry, I read your message wrong.  Please follow the instructions [HERE](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/INSTALL.md) to install kdl, then, when you run rosdep, add `-r` to the end.  If this works for you, I will update the docs for others

baldash, 2021-05-25:
Okay so I installed kdl from sources, cloned geometry2 and ran `rosdep install --from-paths src --ignore-src -ry`, it worked so you can add it to the documentation I think.  
Now the problem is when running `catkin_make_isolated`, I have a build problem occurring with geometry2:  
```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package):
  Could not find a package configuration file provided by "boost_signals"
  (requested version 1.71.0) with any of the following names:

    boost_signalsConfig.cmake
    boost_signals-config.cmake

  Add the installation prefix of "boost_signals" to CMAKE_PREFIX_PATH or set
  "boost_signals_DIR" to a directory containing one of the above files.  If
  "boost_signals" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:443 (find_package)
  CMakeLists.txt:6 (find_package)
```

Which looks like a problem with my boost version I think, I read that they dropped signals in some recent boost update so it must be the problem.

Pratiquea, 2024-08-09:
> I was following the setup tutorial and when running `roslaunch spot_driver driver.launch`
>
> I'm getting this error: `ImportError: cannot import name 'SpotWrapper' from 'spot_wrapper' [...]

spot_wrapper is a package that's included as submodule. So if you had just run the command `git clone https://github.com/heuristicus/spot_ros.git`, you probably didn't get the submodule.

Solution:
`git submodule init`
`git submodule update`

Then go to your workspace and do 
`catkin build`