heuristicus/spot_ros #37: Import errror with tf2_ros on Melodic

harelb, 2021-05-21:
I am having an issue importing tf2_ros in the latest version of the driver. I am running ROS Melodic and I think the issue is tf2_ros is complied for Python 2 which is the default in Melodic.
```Traceback (most recent call last):
  File "/home/.../spot_ros/spot_driver/scripts/spot_ros", line 3, in <module>
    from spot_driver.spot_ros import SpotROS
  File "/home/.../spot_ros/spot_driver/src/spot_driver/spot_ros.py", line 19, in <module>
    import tf2_ros
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
```

I can try to recompile tf2_ros for Python 3 but that breaks other packages. Is there a recommended procedure for installing a Python 3 version of tf2_ros for only this driver? Otherwise, has support for this driver moved to ROS Noetic which defaults to Python 3?

dniewinski, 2021-05-21:
Please follow the install instructions [HERE](http://www.clearpathrobotics.com/assets/guides/melodic/spot-ros/ros_setup.html).  You will need to build a python3 version of geometry for the driver to work.

harelb, 2021-05-21:
Thank you the updated instructions solved this issue. My apologizes, I didn't notice the updated build instructions when I pulled the latest version of the driver.

closed 2021-05-21 by harelb

aamirhatim, 2021-06-10:
The documentation in the provided link should be updated to differentiate system architectures. The `catkin_make` command needs the `DPYTHON_LIBRARY` arg to be `/usr/lib/aarch64-linux-gnu/libpython3.6m.so` if using a Jetson for the backpack PC.