heuristicus/spot_ros #43: Error after claiming the robot

zhaozhongch, 2021-06-19:
After I run the command
```
roslaunch spot_driver driver.launch 
```
I can connect the spot normally and control it through different commands such as sit, stand, velocity.
However, after a few seconds, the terminal will show an error. I can still control the robot, just the error bothers me.
```
Traceback (most recent call last):
  File "/home/zhaozhong/spot/spot_ws/src/spot_ros/spot_driver/scripts/spot_ros", line 7, in <module>
    SR.main()
  File "/home/zhaozhong/spot/spot_ws/src/spot_ros/spot_driver/src/spot_driver/spot_ros.py", line 586, in main
    self.spot_wrapper.updateTasks()
  File "/home/zhaozhong/spot/spot_ws/src/spot_ros/spot_driver/src/spot_driver/spot_wrapper.py", line 402, in updateTasks
    self._async_tasks.update()
  File "/home/zhaozhong/.local/lib/python3.6/site-packages/bosdyn/client/async_tasks.py", line 39, in update
    task.update()
  File "/home/zhaozhong/.local/lib/python3.6/site-packages/bosdyn/client/async_tasks.py", line 90, in update
    self._handle_result(self._future.result())
  File "/home/zhaozhong/.local/lib/python3.6/site-packages/bosdyn/client/common.py", line 458, in result
    raise error
```
The spot software is the latest version

heuristicus, 2021-06-23:
Do you get any specific output from the error? It's raising an exception so there should be some output to the terminal given that the driver doesn't catch it.

civerachb-cpr, 2021-08-10:
Are you still having this problem, or were you able to get it resolved?

civerachb-cpr, 2021-09-02:
Closing this due to inactivity.  Please leave a comment if you're still having this issue.

closed 2021-09-02 by civerachb-cpr