heuristicus/spot_ros #45: tf2 import error when running driver.launch

aamirhatim, 2021-06-30:
I'm running into what looks like to be almost the exact same issue described in #37 after pulling the most recent updates from `master`. This happened with an older version and the solution given in #37 fixed it, but now that fix doesn't seem to work with the latest changes. Any advice on troubleshooting?

Here's the error message:
```
Traceback (most recent call last):
  File "/home/administrator/catkin_ws/src/spot_ros/spot_driver/scripts/spot_ros", line 3, in <module>
    from spot_driver.spot_ros import SpotROS
  File "/home/administrator/catkin_ws/src/spot_ros/spot_driver/src/spot_driver/spot_ros.py", line 19, in <module>
    import tf2_ros
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
```

ROS Melodic
Spot firmware: 2.3.3

civerachb-cpr, 2021-07-08:
The Spot driver uses Python3, which isn't officially supported on Melodic.  It can be made usable though if you run the following:
```
sudo apt-get install python3-pip
pip3 install -U pip
pip3 install rospy
```

Alternatively, you can try changing line 1 of `spot_driver/scripts/spot_ros` from `#!/usr/bin/env python3` to `#!/usr/bin/env python` and see if it works when using the standard Python 2.7 interpreter that ships on Ubuntu 18.04.

civerachb-cpr, 2021-08-10:
I just encountered this same problem installing on 18.04; I was able to get it resolved by simply deleting the build/ devel/ and install/folders from the workspace, ensuring that the geometry2 package was in the workspace, as described on http://www.clearpathrobotics.com/assets/guides/melodic/spot-ros/ros_setup.html#building-the-driver-from-source and building the workspace with

```
catkin_make --cmake-args -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
```

closed 2021-08-10 by civerachb-cpr