heuristicus/spot_ros #48: Build instructions for python3 workaround on ROS Melodic use catkin_make

MarioGini, 2021-09-09:
As the description says, the build instructions suggest to use `catkin_make` when using the python3 workaround for ROS Melodic. I think that in general, using `catkin build` is preferred over `catkin_make` so maybe that section could suggest to e.g. add the additional CMake arguments to a caktin config profile using `catkin config --cmake-args ...`

civerachb-cpr, 2021-09-09:
Realistically you can use either tool, and both will do the job of building the packages.  We tend to use `catkin_make` in a lot of our examples as that command is automatically included (and is also what the ROS wiki uses in its tutorials).

I don't know that there would be a lot of benefit to adding the `catkin build` instructions when the provided instructions using `catkin_make` do work on Melodic.  Melodic is in an awkward situation, since it exists after the Python3 EOL date, but is still stuck without official Python3 support and is stuck requiring these sorts of work-arounds.

peci1, 2021-09-13:
`catkin_make` is the official Melodic tool for building packages. `catkin build` is an alternative implementation. I think it's good the tutorials stick with the officially distributed build system.

civerachb-cpr, 2021-09-13:
I've edited the title of this issue.  Anyone preferring to use `catkin build` can find this issue and use the instructions you've provided here, rather than modifying our core instructions to use a secondary build system.