heuristicus/spot_ros #52: Incomplete PR for Spot arm

EricVoll, 2021-10-01:
Hello everybody,

I made a few commits on my fork ([here](https://github.com/EricVoll/spot_ros/commits/master)) to support camera images from spot's arm and to control the endeffector. I verified that the code works but I can't test the code on a spot without an arm currently, since I won't have access to one for quite some time.

Because of this, I don't want to create a PR on this repo, since it might not work with an "arm-less" spot. 
Would you prefer that I create a WIP PR and you guys can test/edit it, or should I simply do nothing? (I'd love to save somebody some work in the future)

Best,
Eric

civerachb-cpr, 2021-10-18:
I'm not sure what our plans are with regards to getting a Spot with an arm; the Spot we have for development doesn't have the arm, and it's not something you can add afterward.
The best place to add the spot arm code would probably be the Spot Manipulation repo: https://github.com/clearpathrobotics/spot_manipulation -- this is a collection of extra ROS packages for adding manipulators to the base Spot platform.  Because of the tight integration between the base platform and the arm I'm not sure how feasible that would actually be, but on first blush that would be my preference.

EricVoll, 2022-02-22:
I have access to a spot with arm and one without, so I could test the functionality properly now. We'll also need this functionality in the upcoming weeks/months, so we'll do the development anyways. Since the Spot Arm uses the Spot python API (and not moveIt etc.), IMO it would make sense to have in in this repository. If I remember correctly, there are ways in the python package to check if a robot has the arm or not - so the ros wrapper could support both versions.

heuristicus, 2022-02-22:
I agree that it makes sense to have this functionality in one place, since more people are going to have spots with arms.

I have seen that @estherRay has done a significant chunk of work on spot arm integration in her fork at https://github.com/estherRay/spot_ros, and has a separate repository with URDF and so on for the arm at https://github.com/estherRay/Spot-Arm. I don't know how much this overlaps with your PR, but it looks like she has mentioned your work in a few of her commits. I hope that she will consider making a pull request of that work, but she's not made any comments in issues here yet.

estherRay, 2022-02-22:
Hello,
I will make a pull request soon for both the driver and the URDF, moveIt simulation. I also included the code of @EricVoll to visualize the gripper images in my fork. I hope that's still ok to do the pull request.

EricVoll, 2022-02-22:
Hi @estherRay 
A PR would be great. I can review it and test everything. Does your fork contain a service to controll the arm to task-space coordinates? If I remembered correctly, my code to do that was far from perfect and bugged out. Didn't have the time to make that perfect back then (we were at a hackathon...).
If your PR doesn't contain that feature, I'll create a PR with that after your PR.
Thanks!

estherRay, 2022-02-22:
Hi @EricVoll,
My fork contains a service to control the arm by giving a position for each joints. I also tested my code on Spot with an arm and everything works fine for what I've seen. I haven't worked on task-space but it would be great if you manage to immplement that feature!

heuristicus, 2022-10-13:
I'm assuming that #85 does what you've mentioned here and more, so closing.

closed 2022-10-13 by heuristicus