heuristicus/spot_ros #54: Sending trajectory commands very quickly after one has completed sometimes causes the command to stall

heuristicus, 2021-10-12:
I have a node which does a three-stage trajectory, turning, walking, and then turning.

I would occasionally observe that these calls in succession would sometimes result in the robot just not doing anything for however long the duration of the trajectory command I sent was. After some debugging, I noticed that the following section does not check if the trajectory command has unknown status.

https://github.com/clearpathrobotics/spot_ros/blob/8e1554b95cf2875a53f14fda8165957559b0ffb7/spot_driver/src/spot_driver/spot_wrapper.py#L179-L197

I verified that each time I noticed this stall in the trajectory command, the else part of the if statement was being entered. It looks like something is not quite right when sending the goal, but we do not handle this failed case.

I added some sleeps to my three stage command, between the calls to the trajectory actionserver. This seems to have eliminated the problem.

I tested sending commands while a trajectory was still running, and did not notice any problems, so it seems like it might be something to do with a command finishing and sending one right afterwards.

I don't know what the best solution for this is - it's possible that a short (0.5s or less) wait in the method which sends trajectory goals to the wrapper would be sufficient to fix this, and would be transparent to the user, except that if you wanted to interrupt one command with another there would be a short delay.

civerachb-cpr, 2021-10-18:
Thanks for the update on this.  I'll hold off making any changes to this repo for now in case BD gets back to you with a more-correct fix.

Prassi07, 2022-01-23:
Hi, I am trying to use the trajectory commands, and I see this happening frequently. Is there any update on this?

heuristicus, 2022-01-23:
I have received a response from BD but haven't had time to investigate the specific reason it occurs. I have a temporary patch at https://github.com/ori-drs/spot_ros/commit/0f1276a56571b5df8e2623d74dffe3c51980ec5c which I've been using for the last few months which mitigates the issue but does not fix the apparently underlying problem.

Here's what BD told me:

> To get accurate feedback, you have to send the same robot_command_id associated with the specific se2_trajectory request.
> 
> The Clearpath code looks to be sending se2 trajectory commands at about 10Hz and checking feedback at 10Hz.
> 
> At the point when you start receiving STATUS_UNKOWN, it looks like there's a bug in the client where the system that is sending the 10Hz se2 trajectory commands gets restarted. We can see this in the logs where the client sends a command (let's call it cmd #100) and requests feedback on cmd #100. Then 0.01 seconds after cmd #100 is sent, the client starts sending new se2 trajectory commands, cmd #101. However it's still requesting feedback on cmd #100. The trajectory requests and feedback request are thus out of sync at this point.
> 
> If you would modify the feedback to look at feedback.synchronized_feedback.mobility_command_feedback.status, you would see a STATUS_COMMAND_OVERRIDDEN instead of STATUS_PROCESSING.

I want to fix this properly but don't know when I'll have time to investigate exactly why this is happening.

Prassi07, 2022-01-24:
Okayy. I'll keep watch on this until it gets a fix. But as of now, I am just infinitely resending trajectory commands until it receives feedback instead of only sending it once from your temporary patch

heuristicus, 2022-02-25:
I think the cause of this might be related to this part of the code in async_idle

https://github.com/clearpathrobotics/spot_ros/blob/a09e6add6b0ed192ffd60b19c64055d08505d96d/spot_driver/src/spot_driver/spot_wrapper.py#L204-L205

If the idle loop is executed such that somehow the statements that set `is_moving` to true are not entered just after we send a trajectory command, the stand command will be executed, interrupting the trajectory command.

I'm not sure this command at the end of the idle is actually necessary, since the robot should just stand in place if no commands are happening. If we do want to retain this, probably the easiest way to prevent interruption is to check if any of the `_last_*_command` variables are set.

closed 2022-10-18 by heuristicus