heuristicus/spot_ros #57: velodyne networking issue

Chick92, 2021-11-04:
Hi, 

Can you clarify the setup instructions for a spot core and velodyne lidar? i.e the EAP.

I've tried the suggested network configuration, trying to connect directly to 192.168.1.201 and following the instructions on the velodyne ros page - no luck on all front unfortunatly. 

I'm going to reflash the stock image to the EAP and try again, but some clarity would be really helpful. Thanks!

civerachb-cpr, 2021-11-04:
What are the IP addresses of the lidar itself and the computer that's running the ROS nodes?  It's possible that the ROS PC is operating on a different subnet than 192.168.1.x, which would cause problems communicating with the lidar.

[This page](http://www.clearpathrobotics.com/assets/guides/melodic/spot-ros/ros_setup.html) has the default network bridge configuration for the Spot backpack PC:

```
auto lo br0 br0:0 br0:1
iface lo inet loopback

# Bridge together physical ports on machine, assign standard Clearpath Robot IP.
iface br0 inet static
  bridge_ports regex (eth.*)|(en.*)
  address 192.168.131.1
  netmask 255.255.255.0
  bridge_maxwait 0

# Dedicated port for spot
iface br0:0 inet static
  address 192.168.50.1
  netmask 255.255.255.0

# Also seek out DHCP IP on those ports, for the sake of easily getting online,
# maintenance, ethernet radio support, etc.
iface br0:1 inet dhcp
```

Note that the `br0` interface operates on the 192.168.131.x subnet, which corresponds with the default address for the VLP-16 in `spot_driver/launch/accessories.launch`: 192.168.131.20.  You can change the lidar's IP address by setting the `SPOT_VELODYNE_HOST` environment variable in /etc/ros/setup.bash, but it should be on the same subnet as `br0`.

Chick92, 2021-11-04:
I believe the LiDAR is 192.168.1.201 - tried using that and no luck.
The Ip address of the computer is 192.168.1.9 (Lidar subnet) and 192.168.50.5 (spot payload subnet).
I'm using the default spot EAP image, which might be part of the issue as there is a lot of network config already there, and docker seems to be running in the background - this may be for the increaded autonomy (autowalk etc) with the EAP, i dont know as the documentation from BD is not brilliant in this regard. 

I'm downloading the clearpath image now as i think it might be best to do a fresh install and follow your listed instructions EXACTLY, to avoid conflicts.

civerachb-cpr, 2021-11-04:
If you're using Ubuntu 18.04 we've noticed that frequently multiple static interfaces on a bridge defined with the /etc/network/interfaces file don't always come up correctly on boot.

If you reboot the machine, log in, and immediately run `ifconfig -a` and see that you don't have all of the static bridges up then I'd suggest adding `ifup <interface>` to `/etc/rc.local` to force them up when the computer starts.

closed 2022-10-13 by heuristicus

civerachb-cpr, 2023-03-09:
What model of lidar sensor are you using?  Provided the lidar is configured to use a static IP address on the same subnet as `eno1` and you're able to ping the lidar from the SpotCore PC, there shouldn't be any issues getting the lidar data into ROS using the appropriate driver for that sensor.

In general, you will need to:
1. Configure the lidar to use a static IP address on the same subnet as `eno1`.  Depending on the exact sensor you may also need to set other configuration options on the sensor itself; consult the sensor's manual for guidance here.
2. Connect the lidar to `eno1`.  Ensure you can ping the lidar's IP address from the SpotCore.
3. Start the ROS node(s) for the lidar driver.  This will vary depending on the exact model you're using. For example, if it's a VLP-16 you could use `roslaunch velodyne_pointcloud VLP16_points.launch`, specifying the lidar's IP address as an argument to the launch file.
4. Start gmapping, specifying the 2D laserscan topic you want to use as the input the SLAM system.

Since this threat was originally regarding the Velodyne VLP-16, I'm going to assume that's the same sensor you're using?  By default the `velodyne_pointcloud` package's launch file for that sensor is configured to publish a 2D `sensor_msgs/LaserScan` topic, normally called `/scan`.  You should be able to configure `gmapping` to read sensor data from this topic.  You'll need to ensure your lidar sensor frame is included in Spot's URDF, so that the `tf` from the sensor frame to the robot's `base_link` can be calculated.

Gmapping on Spot with the VLP-16 will have a few difficulties which may hamper performance:
- Because Spot's body moves up and down while walking, the 2D laserscan data won't always be on the same plane.  This will likely cause some additional noise and may make localization more difficult.
- Gmapping _only_ works with 2D lidar data, so the robot will not be able to see any obstacles above/below the plane of the lidar.  This may result in gmapping planning paths that will collide with low obstacles that are invisible to the lidar
- The VLP-16's minimum range is about 1m.  Anything closer than this distance to the robot will be invisible to the lidar.  In practice this means that sometimes gmapping will move too close to an obstacle, such that it's now within the minimum range of the sensor. This means gmapping will no longer be aware that this obstacle exists, and will make no effort to navigate around it.

danchoy, 2023-03-09:
Thanks for your reply.

The lidar in use is VLP16, which is also deliver by Boston Dynamics. 
My objective is to connect the lidar from my desktop computer that already has ROS Melodic installed via two connections - Ethernet for internet and Wi-Fi network for the Spot access point.

However, the current issue is that there is no way to establish a connection between the lidar and my desktop computer. While I can control the Spot using the Spot_Driver from my desktop, I cannot communicate with the lidar due to network settings.

My Desktop Computer network:
Ethernet: 10.0.0.x connection for internet 
Wifi: 192.168.80.x connection to the Spot 

Spot (the robot)
Access Point: 192.168.80.3
Spot Core connection: 192.168.50.3

Spot Core 
Ethernet 1 : 192.168.1.9 (connection for the lidar)
Ethernet 2 : 192.168.50.5 (connection for the Spot)

The lidar (VLP16)
Ethernet : 192.168.1.201 (connection to Spor Core)

I hope the information provided above can assist you in identifying the source of my issue.

Thank you.

Chick92, 2023-03-09:
Use wireshark, or tshark if via CLI to find the velodynes IP. configure the VLP16 launch file to use that IP, and create a ros multi machine network. Then you can get the lidar frames on the remote PC. 

The lidar and payload ethernet ports are not bridged by default on the EAP1.

You can change the velodyne's IP to something else, but then the autowalk service won't work and the robot will complain that it can'tfind the lidar.

danchoy, 2023-03-10:
> Use wireshark, or tshark if via CLI to find the velodynes IP. configure the VLP16 launch file to use that IP, and create a ros multi machine network. Then you can get the lidar frames on the remote [...]

Thanks for your reply.

I have obtained the IP address for the LiDAR, but I am facing an issue in bridging it with my desktop computer. Although I don’t mind to install another ROS in the Spot Core, it will only enable access to the LiDAR directly and not allow publishing of the topic outside the Spot Core.

Chick92, 2023-03-10:
Install ros on the core

Install ros on the other computer

Setup ros multi machine network 

get velodyne frames on other computer - that is the only way to do what you're asking.

danchoy, 2023-03-10:
> Install ros on the core
> 
> Install ros on the other computer
> 
> Setup ros multi machine network
> 
> get velodyne frames on other computer - that is the only way to do what you're asking.

Thanks, let me take a look how does it work

Chick92, 2023-03-10:
Thinking about it, your setup will need a bit more work because you're using wifi rather than a dedicated radio via ethernet.

use a wifi dongle with the core and connect to spot's AP

add these to the bashrc files respectively 

CORE
export ROS_MASTER_URI=http://192.168.80.**:11311
export ROS_IP=192.168.80.**

Computer
export ROS_MASTER_URI=http://192.168.80.**:11311
export ROS_IP=192.168.80.&&

Where ** is the subnet of the core and && is the subnet of your computer

civerachb-cpr, 2023-03-10:
Technically you don't actually need to establish communication between the VLP-16 and your laptop; the ROS node for the Velodyne should be running on the SpotCORE.  The LaserScan and PointCloud2 messages will be published by the SpotCORE, so as long as you have communication between the SpotCORE and your laptop you'll be able to receive the lidar data remotely.

Something like this should work for what you're describing:
```
+-----------+ enp2s0 +----------+  eno1  +--------+
| Spot Base | ------ | SpotCORE | ------ | VLP-16 |
+-----------+        +----------+        +--------+
                          |
                          | wifi
                          |
                      +--------+
                      | Laptop |
                      +--------+
```

The SpotCORE will be the ROS master for the system, running the `spot_ros` driver (to control the robot) & the velodyne driver.  Your laptop should be configured to use the SpotCORE as its ROS_MASTER_URI (see @Chick92's comments above on how to set up a remote ROS master).

The Gmapping nodes can either be run on your laptop, communicating with the SpotCORE over wireless, or you can run gmapping on the SpotCORE itself and simply use your laptop to run something like Rviz to monitor the robot.  (Personally I prefer running gmapping on the SpotCORE since that will introduce less latency to the SLAM.)

danchoy, 2023-03-10:
> 

Thanks for your advice. 
I think your diagram might be the easiest way to achieve what I want to do. 
If I need to set up the environment as shown in your diagram, then I will have to use a wifi dongle to connect the Spot Core to the same subnet as my computer.

However, my original plan try not involve using a USB wifi dongle, as I'm concerned that it might not have enough throughput for the data transfer.

+-----------+ enp2s0 +----------+  eno1  +--------+
| Spot Base | ------ | SpotCORE | ------ | VLP-16 |
+-----------+        +----------+        +--------+
     |
     | wifi
     |
     +--------+
   | Laptop |
     +--------+

civerachb-cpr, 2023-03-10:
The bandwidth requirements really depend on what exactly you're doing. Prior to handing over ownership of the `spot_ros` repo to its current maintainers, Clearpath Robotics used the setup I described above for all of the ROS-enabled Spot robots we integrated; we used a USB wireless dongle regularly and never had bandwidth issues with any of the robots we set up.

The biggest consumers of bandwidth will be the camera & lidar data.  But if you aren't subscribing to those topics from your laptop (or at least not all of them concurrently) you should be able to get away with fairly modest wireless bandwidth.  That's why I recommended running Gmapping directly on the SpotCORE and using your laptop just for displaying the robot's position on the map using Rviz; in that sort of a setup the raw lidar messages never need to get transmitted over the wireless; they're consumed locally by Gmapping on the SpotCORE, and all that's sent wirelessly is the map and TF data.  That leaves plenty of bandwidth to subscribe to a camera feed or even the laserscan data if you need to visualize it.

danchoy, 2023-03-10:
> 

Unfortunately, my plan is to use my laptop to run gmapping and subscribe to the lidar topic via SpotCORE. Let's see if the wifi dongle can handle it. If not, I might have to run gmapping directly from SpotCORE.

danchoy, 2023-03-10:
> The bandwidth requirements really depend on what exactly you're doing. Prior to handing over ownership of the `spot_ros` repo to its current maintainers, Clearpath Robotics used the setup I [...]

Unfortunately, my plan is to use my laptop to run gmapping and subscribe to the lidar topic via SpotCORE. Let's see if the wifi dongle can handle it. If not, I might have to run gmapping directly from SpotCORE.