heuristicus/spot_ros #61: Navigation Action issue

edomil90, 2022-02-04:
Hey guys,

I had several issues using the /spot/navigation_to action. Did someone use it successfully? First of all, the upload of the graph_nav takes a lot of time, roughly 7s per waypoint, while on the sdk the whole upload takes few seconds. Anyway, the action always stops due to some errors in the code, and even though I fixed some of those errors (I can make a PR with my corrections so far), new ones are coming up. The last one, after the initial fiducial was successfully detected:

[ERROR] [1643984336.793953]: Exception in your execute callback: not all arguments converted during string formatting
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/actionlib/simple_action_server.py", line 289, in executeLoop
    self.execute_callback(goal)
  File "/home/spot/catkin_ws/src/spot_ros/spot_driver/src/spot_driver/spot_ros.py", line 443, in handle_navigate_to
    initial_localization_waypoint = msg.initial_localization_waypoint)
  File "/home/spot/catkin_ws/src/spot_ros/spot_driver/src/spot_driver/spot_wrapper.py", line 671, in navigate_to
    self._list_graph_waypoint_and_edge_ids()
  File "/home/spot/catkin_ws/src/spot_ros/spot_driver/src/spot_driver/spot_wrapper.py", line 738, in _list_graph_waypoint_and_edge_ids
    graph, localization_id, self._logger)
  File "/home/spot/catkin_ws/src/spot_ros/spot_driver/src/spot_driver/graph_nav_util.py", line 109, in update_waypoints_and_edges
    edge.id.to_waypoint)
  File "/usr/lib/python3.6/logging/__init__.py", line 1308, in info
    self._log(INFO, msg, args, **kwargs)
  File "/usr/lib/python3.6/logging/__init__.py", line 1444, in _log
    self.handle(record)
  File "/usr/lib/python3.6/logging/__init__.py", line 1454, in handle
    self.callHandlers(record)
  File "/usr/lib/python3.6/logging/__init__.py", line 1516, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.6/logging/__init__.py", line 865, in handle
    self.emit(record)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosgraph/roslogging.py", line 245, in emit
    record_message = _defaultFormatter.format(record)
  File "/usr/lib/python3.6/logging/__init__.py", line 577, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.6/logging/__init__.py", line 338, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting

This looks strange because it should be a copy and paste of the sdk example of graph_nav, so what could be wrong? Thanks for helping!

civerachb-cpr, 2022-02-04:
Sorry, I haven't had a chance to dig into any of these problems.  Our development Spot is currently down for repairs, which has stalled work a lot of the issues that have been reported.  Someone else in the community may have had more success than me though.

edomil90, 2022-02-07:
I used this fork for the graph_nav navigation and it works just great! 

https://github.com/sktometometo/spot_ros