heuristicus/spot_ros #62: Arm and gripper

zkytony, 2022-02-09:
Is there a way to obtain the URDF for arm and gripper? Also, obtain images from the gripper's camera? using the spot_ros stack?

heuristicus, 2022-02-10:
There isn't currently a URDF for the arm/gripper. If you have a robot with an arm perhaps you can ask boston dynamics to provide you with .dae files for the arm components and gripper. You can see the current URDF at https://github.com/clearpathrobotics/spot_ros/tree/master/spot_description.

For the camera, you should be able to use a similar approach to

https://github.com/clearpathrobotics/spot_ros/blob/a09e6add6b0ed192ffd60b19c64055d08505d96d/spot_driver/src/spot_driver/spot_wrapper.py#L240-L242

and 

https://github.com/clearpathrobotics/spot_ros/blob/a09e6add6b0ed192ffd60b19c64055d08505d96d/spot_driver/src/spot_driver/spot_wrapper.py#L289-L291

heuristicus, 2022-02-10:
If you get those things please make a pull request - it would be a useful addition to the driver.

heuristicus, 2022-02-10:
Actually, I just noticed that @estherRay has been working on adding spot arm commands to the driver on her branch https://github.com/estherRay/spot_ros, and has a separate repository with URDF and so on for the arm at https://github.com/estherRay/Spot-Arm.