heuristicus/spot_ros #71: tf transform between 'odom' and 'body' seems wrong

zkytony, 2022-03-05:
In the screenshot below,
[image]

The robot is sitting on a dock, but the height difference between the odom frame and the body frame is too much. 
I have tried restarting the driver, and the height difference would vary (there are times when the height seems to be reasonable, but it is very rare). What could be the reason?

heuristicus, 2022-03-05:
This is caused by odometry drift. If you reboot the robot you should see that the odometry starts zeroed. If you move the robot around, the odometry will drift in the x-y coordinates. But, the z-offset drifts regardless of whether the robot is in motion, which I suspect is caused by IMU drift.

You should not rely on odometry to provide you with an accurate position of the robot. Over short periods of time the estimate is good, but over long periods is is unreliable.

zkytony, 2022-03-05:
The thing is It didn't move the robot around, and it would start with an incorrect height (the robot is just staying on the dock the whole time).

closed 2022-03-06 by zkytony

zkytony, 2022-03-09:
setting the parameter `mode_parent_odom_tf` to `vision` helped!