heuristicus/spot_ros #77: Vision Frame drift

dberrett, 2022-05-20:
How are folks relocalizing Spot?  When running Spot for a while, my Vision/Odom frames get radically offset from the robot.  I.e. the Vision frame is 15 meters below Odom frame, and the robot's GPE is another 10 meters below and away from that.  So when building my map and costmaps, everything is offset and the heights aren't valid.
I was assuming that the Vision frame would be accurate over time, but maybe I need something like AMCL instead?

Unless I'm just missing something obvious?

heuristicus, 2022-05-20:
This might be something that would be good to ask on the BD forums or support if you have access to that. In practice the easiest way to reset everything is to reboot the robot, although that does seem like it should be unnecessary.

If you're using the pose output from the driver, one way of compensating might be to write something which saves the current values of all the frames you're interested in, and then publishes the offsets from those saved frames to the live values from the robot. I'm not sure if that's sensible in practice but I've thought about doing something like that.

dberrett, 2022-05-20:
Thanks for the suggestions.  I'll poke around with the ideas and see if I can get something to work.

heuristicus, 2022-05-20:
For reference, the place where tf frames are generated is here
https://github.com/clearpathrobotics/spot_ros/blob/a09e6add6b0ed192ffd60b19c64055d08505d96d/spot_driver/src/spot_driver/ros_helpers.py#L313-L340

closed 2022-10-13 by heuristicus