heuristicus/spot_ros #86: Maintenance - Is this repo active?

EricVoll, 2022-10-13:
Hi @dniewinski and others,

question: Is this repo actively maintained? The last commits to master are pretty old, and many open PRs could be merged.

Best
Eric

EricVoll, 2022-10-13:
Hi @heuristicus 
thanks for the info. 
I work at Microsoft now and we're using Spot with this ros wrapper quite actively in different projects. I can ask internally if there is an option for Microsoft's GitHub Account to take over this repository (e.g. as a fork?).

I agree with you @heuristicus that we should keep this repository alive in some way.

heuristicus, 2022-10-13:
As long as there's someone willing to actively maintain it I'm not attached to it going to any place specific. I spoke to people at my institute and there was no objection in principle to moving it to the https://github.com/ori-drs organisation. We are using it here for our spot, but it's only really me making use of it. It probably makes sense for you or your team to be the organisational host if you have more resources available.

EricVoll, 2022-10-17:
I think our resources are usually limited for non-prod work, so I'd be super glad if you could host it on the ori-drs account, that would be awesome. 

I can see the repo has been assigned to your account already (confused me for a second there :D)

heuristicus, 2022-10-17:
OK, I'll make that change at some point soon. But at any rate now we can start merging stuff again.

closed 2022-10-17 by heuristicus