heuristicus/spot_ros #96: Velocity Commands behave weird on newest Spot SDK

EricVoll, 2022-10-27:
I observed that the /spot/cmd_vel topic behaves super weird on the newest spot sdk. Spot would start to move slightly, stop, continue etc. 
Increasing the command duration from the current 125ms to 400ms mitigates the problem (but it's still not completely gone). After downgrading spot to an older version, the problem is gone and the current cmd_vel implementation works fine. Note: the problem is not the version of the python pip package, but the actual spot version.

I don't currently know which versions our spots have exactly, but I'll figure that out and update this issue.

heuristicus, 2022-10-28:
It seems like it might be a recurrence of #54. At least, it sounds superficially similar in terms of the stop/start behaviour. What rate are you sending the commands at?

Might be worth checking the details in the documentation for the velocity command we're using and see if it's changed at all in recent updates.

jeremysee2, 2023-04-03:
I've been testing the cmd_vel topic with Spot v3.2.0, found that it works better if you increase the command duration to 600ms. This follows the [wasd example from the SDK](https://github.com/boston-dynamics/spot-sdk/tree/master/python/examples/wasd) where they use 0.6s command duration.

https://github.com/boston-dynamics/spot-sdk/blob/208c885d99813d02b4d78940289cf1c9d0279674/python/examples/wasd/wasd.py#L46-L49

This example uses the same `RobotCommandBuilder.synchro_velocity_command` proto that this package uses for sending velocity commands, so its safe to follow the same command duration. I've tested it to work on my Spot.

commit 84a739ab (2023-04-07): increase velocity command timeout to 0.6 seconds

probably fixes #96
closed 2023-04-04 by heuristicus