heuristicus/spot_ros #99: ImportError: dynamic module does not define module export function (PyInit_PyKDL)

BCaran, 2022-11-11:
Hi,

when I start:

`roslaunch spot_driver driver.launch`

```
Traceback (most recent call last):
  File "/home/spot/catkin_ws/src/spot_ros/spot_driver/scripts/spot_ros", line 3, in <module>
    from spot_driver.spot_ros import SpotROS
  File "/home/spot/catkin_ws/src/spot_ros/spot_driver/src/spot_driver/spot_ros.py", line 23, in <module>
    import tf2_geometry_msgs
  File "/home/spot/catkin_ws/devel/lib/python3/dist-packages/tf2_geometry_msgs/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/spot/catkin_ws/src/geometry2/tf2_geometry_msgs/src/tf2_geometry_msgs/tf2_geometry_msgs.py", line 31, in <module>
    import PyKDL
ImportError: dynamic module does not define module export function (PyInit_PyKDL)
```

heuristicus, 2022-11-14:
You're probably missing the `python3-pykdl` package. You can install it with apt.

closed 2022-11-14 by heuristicus

BCaran, 2022-11-14:
When I try to install it with sudo apt, I got error that there is no python3-pykdl.

BCaran, 2022-11-14:
I fixed this issuse with installing Ubuntu 20.04 for Jetson Xavier AGX and installing ROS Noetic.

heuristicus, 2022-11-14:
If you were not using noetic before, that may have been another part of the problem - this package does not support melodic or older versions of ROS.

BCaran, 2022-11-14:
I did some research and problem is that PyKDL is now  merged together for Python2 and Python3.

Usage of Noetic works like charm because Python 3 is official interpreter.

danchoy, 2023-03-09:
To resolve this problem, I utilized Ubuntu 18.04 with Python versions 3.69 and 2.7. 
It was necessary to manually install all required modules. 
Additionally, the current version of orocos_kdl on Github was not compatible with my environment, 
so I opted to clone the previous version.

cd ~/catkin_ws/src
sudo apt update
sudo apt install libeigen3-dev libcppunit-dev
sudo apt install python3-psutil python3-future
python3 -m pip install psutil
python3 -m pip install future
git clone -b release-1.5 https://github.com/orocos/orocos_kinematics_dynamics.git
cd orocos_kinematics_dynamics/python_orocos_kdl
git submodule update --init
catkin build orocos_kdl
catkin build python_orocos_kdl