heuristicus/spot_ros #100: RGB8 images are marked as big endian, why?

peci1, 2023-01-04:
https://github.com/heuristicus/spot_ros/blob/bb82c29e9d51d12d57d53c8b4d4faecb600a1bc8/spot_driver/src/spot_driver/ros_helpers.py#L150-L154

and similar are present in the camera driver part. The only little-endian thing is the 16UC1 depth image (which is correct as protobufs are always little).

Is there some reason for the single-byte values to be marked as big endian? If I understand it correctly, this has nothing to do with BGR/RGB as that is one level higher than endiannes...

heuristicus, 2023-01-05:
I don't know much about how ROS makes use of the endianness values when processing/publishing, but I suppose this might be because some things wouldn't work correctly if the endianness is left on its default of false?

Perhaps it's something to try removing and seeing if the images are still output as expected in rviz or other image viewers.

peci1, 2023-01-12:
I've tested it with both the mono images from body cameras and with the RGB camera from arm. Nothing changed in the look of the images as viewed by image_view, rqt_image_view, rviz or rqt_bag.