heuristicus/spot_ros #117: Error when starting the driver on ROS Melodic on Ubuntu 18.04.5 LTS on a SpotCORE

Vugul, 2023-03-31:
Hello, I'm using a Spot Core to get the Spot_ROS driver running to control the Spot. The Core is directly connected to the Spot. My standard Python Version is Python 2.7.17. 
When I'm trying to launch the driver I get the Error messages in the pictures down below. I already tried the "Building for Melodic" chapter but it did not help. 

It would be really awesome if somebody could help me solve the problem.
Thank you so much in advance.

Kind regards,
Vugul

[image]
[image]

Vugul, 2023-03-31:
Thank you for your help. I will try those steps and see if it works.

Is it possible to just use a virtual environment with python3 and install the Spot Ros Package with Python3 running?
When I try to test the view_model.launch in a Python3 Virtual Environment I get an error that the joint_state_publisher has died and the joints arent visualized in RVIZ.
If it would be possible to install it on a Python3 Environment, I could maybe try to avoid the other problem.

[image]
[image]

Vugul, 2023-04-06:
Would it be possible to delete ROS melodic fully from the SpotCore and install it with a python3 Virtual Environment running to make it only use Python3 and because of this the error from above won´t happen?

heuristicus, 2023-04-07:
I'm not sure how the spot CORE is configured - based on the BD documentation it seems like it's 18.04 only and there's likely some specific configuration associated with that. As I said above, you're probably best off running noetic in a docker container.

closed 2023-06-03 by heuristicus