heuristicus/spot_ros #131: OpenSSL Handshake failed

MarioSpeckert, 2023-05-02:
Hi,
this might not be related to this Repository, but we have a connection issue to the Spot robot.
It seams that the SSL handshake is not working due to some certification problem. 
Are there additional steps needed for setting up the connection?
Please let us know if you need more Information.

Regards
Mario

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal

roslaunch spot_driver driver.launch 
... logging to /home/mario/.ros/log/7096e4c6-e901-11ed-b88c-13df06778170/roslaunch-mario-ThinkPad-T430s-46920.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mario-ThinkPad-T430s:45537/

SUMMARY
========

PARAMETERS
 * /bluetooth_teleop/joy_node/autorepeat_rate: 20
 * /bluetooth_teleop/joy_node/deadzone: 0.1
 * /bluetooth_teleop/joy_node/dev: /dev/input/js0
 * /bluetooth_teleop/teleop_twist_joy/axis_angular/yaw: 3
 * /bluetooth_teleop/teleop_twist_joy/axis_linear/x: 1
 * /bluetooth_teleop/teleop_twist_joy/axis_linear/y: 0
 * /bluetooth_teleop/teleop_twist_joy/enable_button: 4
 * /bluetooth_teleop/teleop_twist_joy/enable_turbo_button: 5
 * /bluetooth_teleop/teleop_twist_joy/scale_angular/yaw: 1.4
 * /bluetooth_teleop/teleop_twist_joy/scale_angular_turbo/yaw: 1.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear/x: 0.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear/y: 0.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear_turbo/x: 2.0
 * /bluetooth_teleop/teleop_twist_joy/scale_linear_turbo/y: 2.0
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /spot/spot_ros/allow_motion: True
 * /spot/spot_ros/auto_claim: False
 * /spot/spot_ros/auto_power_on: False
 * /spot/spot_ros/auto_stand: False
 * /spot/spot_ros/autonomy_enabled: True
 * /spot/spot_ros/claim/get_lease_on_action: False
 * /spot/spot_ros/claim/use_take_lease: False
 * /spot/spot_ros/estop_timeout: 9.0
 * /spot/spot_ros/hostname: xxx.xxx.xxx.xxx
 * /spot/spot_ros/max_angular_velocity_z: 0
 * /spot/spot_ros/max_linear_velocity_x: 0
 * /spot/spot_ros/max_linear_velocity_y: 0
 * /spot/spot_ros/password: xxxxxxx
 * /spot/spot_ros/rates/check_subscribers: 10.0
 * /spot/spot_ros/rates/feedback: 10.0
 * /spot/spot_ros/rates/front_image: 10.0
 * /spot/spot_ros/rates/hand_image: 10.0
 * /spot/spot_ros/rates/lease: 1.0
 * /spot/spot_ros/rates/loop_frequency: 50.0
 * /spot/spot_ros/rates/metrics: 0.04
 * /spot/spot_ros/rates/mobility_params: 10.0
 * /spot/spot_ros/rates/point_cloud: 10.0
 * /spot/spot_ros/rates/rear_image: 10.0
 * /spot/spot_ros/rates/robot_state: 20.0
 * /spot/spot_ros/rates/side_image: 10.0
 * /spot/spot_ros/username: admin
 * /twist_marker_server/linear_scale/x: 1
 * /twist_marker_server/linear_scale/y: 1
 * /twist_marker_server/link_name: body
 * /twist_marker_server/max_negative_linear_velocity/x: -1
 * /twist_marker_server/max_negative_linear_velocity/y: -1
 * /twist_marker_server/max_positive_linear_velocity/x: 1
 * /twist_marker_server/max_positive_linear_velocity/y: 1
 * /twist_marker_server/robot_name: spot
 * /twist_mux/locks: []
 * /twist_mux/topics: [{'name': 'bt_joy...

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    twist_marker_server (interactive_marker_twist_server/marker_server)
    twist_mux (twist_mux/twist_mux)
  /bluetooth_teleop/
    joy_node (joy/joy_node)
    teleop_twist_joy (teleop_twist_joy/teleop_node)
  /spot/
    spot_ros (spot_driver/spot_ros)

auto-starting new master
process[master]: started with pid [46937]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7096e4c6-e901-11ed-b88c-13df06778170
process[rosout-1]: started with pid [46950]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [46957]
process[twist_marker_server-3]: started with pid [46958]
process[bluetooth_teleop/joy_node-4]: started with pid [46959]
process[bluetooth_teleop/teleop_twist_joy-5]: started with pid [46963]
process[spot/spot_ros-6]: started with pid [46971]
process[twist_mux-7]: started with pid [46978]
[ WARN] [1683042792.518570003]: Couldn't set gain on joystick force feedback: Bad file descriptor
[INFO] [1683042793.327260]: Starting ROS driver for Spot
[INFO] [1683042793.330749]: Initialising robot at xxx.xxx.xxx.xxx
[INFO] [1683042793.332368]: Trying to authenticate with robot...
E0502 17:53:13.344975498   46971 ssl_transport_security.cc:1420]       Handshake failed with fatal error SSL_ERROR_SSL: error:1000007d:SSL routines:OPENSSL_internal:CERTIFICATE_VERIFY_FAILED.
[WARN] [1683042793.346742]: Failed to communicate with robot: UnableToConnectToRobotError: The robot may be offline or otherwise unreachable.
Ensure the robot is powered on and you can ping xxx.xxx.xxx.xxx. Robot may still be booting. Will retry in 15 seconds
E0502 17:53:23.341092687   47020 ssl_transport_security.cc:1420]       Handshake failed with fatal error SSL_ERROR_SSL: error:1000007d:SSL routines:OPENSSL_internal:CERTIFICATE_VERIFY_FAILED.
[INFO] [1683042808.362866]: Trying to authenticate with robot...
[WARN] [1683042808.366667]: Failed to communicate with robot: UnableToConnectToRobotError: The robot may be offline or otherwise unreachable.
Ensure the robot is powered on and you can ping xxx.xxx.xxx.xxx. Robot may still be booting. Will retry in 15 seconds

GnomeZone, 2023-05-03:
Can you successfully control the robot with the controller? What version of the spot SDK have you got installed?

heuristicus, 2023-05-03:
This isn't something I've seen before. One thing to test would be to try and connect to the robot with this example provided by BD: https://github.com/boston-dynamics/spot-sdk/blob/master/python/examples/hello_spot/hello_spot.py. If that doesn't work I'd recommend contacting BD support to try and work out what the problem is.

MarioSpeckert, 2023-05-03:
Apparently there is a problem with the laptop I am using. With another computer, the connection works perfectly. Nevertheless, thanks for the help.

closed 2023-05-03 by MarioSpeckert