heuristicus/spot_ros #135: Several nodes not loading on driver launch

Sibren202, 2023-06-08:
When launching the driver I get several error of nodes not loading, including twist, cmd_vel, and more. After running the rviz I am able to succesfully power on and claim lease, however sending messages to these topics does not work and gives a time-out error in the terminal. Im trying to figure out why launching the driver does not verify the nodes. Have re cloned and built the git a thousand times already but cant seem to find the issue...

heuristicus, 2023-06-08:
What version of the driver do you have on the robot, and what version of the SDK on the computer you're running the driver with?

Sibren202, 2023-06-08:
Sorry should have put that in there first time.
Think I'm using 3.2.3 for both? Could a version mismatch be the cause of the error?

heuristicus, 2023-06-08:
It's possible that if you were using a really old version it wouldn't work, but if you're on anything above version 3 it should be fine.

What do you mean by a timeout error? Is it an error from the driver itself, ROS, or an error that's forwarded from the SDK? Are you able to use sit/stand/self-right/roll over and so on? If you use the rviz interface, can you use the 2d nav goal to move the robot?

There have been a few reports of issues with cmd_vel https://github.com/heuristicus/spot_ros/issues/96, but it seemed to be an issue with timings which I fixed. Perhaps the fix didn't actually work.

heuristicus, 2023-06-08:
Perhaps related to #74?

Sibren202, 2023-06-08:
[ERROR] [1683799282.850639]: Unable to execute robot command: bosdyn.api.RobotCommandResponse (ExpiredError): The command was received after its max_duration had already passed.

This the exact error when I try to send a message to cmd_vel or twist. Sending to go_to_pose works fine and Spot moves. And yes 2d nav goals in rviz works.