heuristicus/spot_ros #142: TF_REPEATED_DATA ignoring data with redundant timestamp

mrayh001, 2023-08-16:
Hi, 

I am running the driver on Ubuntu 20.04 and ROS noetic. 

spot_ros: 58ba355a7938f2b2d9bd062638e3efc9484f2e96
spot_wrapper: d96bdbf41445215500fb6bc8e7ae278bb1957746

I see the spam on my terminal [ WARN] [1692223451.391583276]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame dock_521 at time 1692223451.000484 according to authority unknown_publisher

and Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame dock_521 at time 1692223520.812809 according to authority /spot/spot_ros

Is there any solution to this? Thanks

trangml-mit, 2023-12-06:
Is there any update on the fix to this? I'm not too sure where a list of published tfs could be kept to avoid this problem.

The constant print spam makes it pretty difficult to debug other messages.

heuristicus, 2023-12-07:
I attempted a fix at https://github.com/heuristicus/spot_ros/blob/2a1d26da976a6376b21b5af9ce5328899fb7a8c4/spot_driver/src/spot_driver/spot_ros.py#L162-L178 a while back, I suppose it's still missing some of the duplicates.

trangml-mit, 2023-12-19:
Yeah, I realize now that it's actually having a problem due to any detected fiducials, so those may need special error handling. For now, my fix is just covering the fiducial.
```

Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fiducial_520 (parent vision) a
t time 1703005423.525936 according to authority /spot/spot_ros                                           
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp                          
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame filtered_fiducial_520 (parent 
vision) at time 1703005423.525936 according to authority /spot/spot_ros                                  
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp                          
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame dock_520 (parent vision) at ti
me 1703005423.525936 according to authority /spot/spot_ros                                               
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp                          
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame dock_520 (parent vision) at ti
me 1703005423.525936 according to authority /spot/spot_ros                                               
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp
```