heuristicus/spot_ros #143: Spot Arm URDF appears to be faulty

EceChaik, 2023-09-20:
Hi,

It seems that, at least in our lab and for our Spot robot, the spot_arm urdf file provided, does not match the real system.
For example, Boston Dynamics provides a tf publisher for the position of the hand relative to the body, which corresponds to the real-world location of the end-effector, but is very different to the one predicted by the urdf when visualized in rviz.
Also, in terms of simulation, the mass values provided for the links are too large, the real robot arm weighs close to 5Kg in total.
I have created a modified xacro file, which at least in our case, appears to be the correct one and when used with the robot, the urdf is aligned with the locations published by Boston Dynamics.
Let me know if I should do a pull request in case it is of interest. Or if I am the only one who has run into this issue.

heuristicus, 2023-09-20:
I don't have access to a spot with an arm so can't verify any of the arm components, though they were all submitted by people who did have access to the arm. It would be great if you made a PR with your changes, and have some videos of the problem before and after the changes.

EceChaik, 2023-09-20:
Just checked the spot_ros2 implementation you provided, clearly there are big differences compared to the urdf hosted here. Perhaps I'll find the time to use it with the real robot soon, then upload some videos with all the different urds.