heuristicus/spot_ros #145: Unable to get data from payload camera

ajay1606, 2023-10-20:
Hello,

I am using ROS driver for SPOT control and able to get data from the default cameras installed in the SPOT.
But with the additional Payload cameras (  [Spot Cam+IR](https://support.bostondynamics.com/s/article/Spot-CAM-Spot-CAM-Spot-CAM-IR)), unable to get thermal image topics in the ROS node.

Able to get default camera (installed on SPOT), but not the payload camera data.
[image]

But in the controller display i am able to see the payload camera data, but not able to get those data in the ROS node.
[image]

Appreciate any suggestions !

Development Platform: Ubuntu 20.04
ROS : Noetic

heuristicus, 2023-10-23:
Hi, please check the information at https://heuristicus.github.io/spot_ros/html/cam_usage.html about how to use the spot cam with this driver. It seems like you may not be running the additional launch file.

ajay1606, 2023-10-23:
@heuristicus Thank you so much for your quick response. And I have gone through the link you have shared. After installing **aiortc==1.3.2** and i have tried to lauch Spot cam with following command, with our password.

`roslaunch spot_cam spot_cam.launch username:=user password:=spot_password hostname:=192.168.80.3`
But end up with error message like bellow. Would you please help us to fix this .

[image]

heuristicus, 2023-10-24:
Please try again with the latest commit I just made. It looks like there was an issue with the import name not being correct.

ajay1606, 2023-10-24:
@heuristicus thanks for your response after tried with latest commit changes also got same error . Any how after changing same thing in here also, am able to get **spot_cam node** launch normally.
https://github.com/heuristicus/spot_ros/blob/a76ded46bdeaa9ac3f054a1b0f4facd90adc3f74/spot_cam/src/spot_cam/spot_cam_ros.py#L328

But unfortunately, after spot_cam node started, am getting following message in the terminal like below, still unable to visualize thermal image in rostpoic. 

[image]

Should I change IP in the launch file ? currently its set to default "192.168.50.3"

closed 2023-10-24 by ajay1606

heuristicus, 2023-10-25:
The output of the camera image topic is controlled by setting the screen that is displayed, which can be done with the `/spot/cam/set_screen` topic. See https://heuristicus.github.io/spot_ros/html/cam_usage.html#screens for more details.

ajay1606, 2023-10-25:
@heuristicus Thank you so much for your continuous response, now its working well. But there is no any topic publishing camera info for `/spot/cam/info` ?

And also, data currently publishing at 3 hz, is there any parameters to tune ?