heuristicus/spot_ros #146: 'catkin build' or 'catkin_make'

grapesonwang, 2023-11-09:
Hi guys,

I am using spot_ros to work with Spot Dog and have some questions as below. I am using 'noetic + Ubuntu 20.04LTS'

**Question 1: Should I use 'catkin build' or 'catkin_make' to build spot_ros?** 

The reasons are: I can successfully build spot_ros using 'catkin build', but when I use 'catkin_make', there are some errors as below. So would this really matter? I understand 'catkin build' resolves dependency better than 'catkin_make', so I assume this is due to spot_viz being built before the needed message/service header files were created. Please let me know if this is the cas.

`~/spot_ws$ catkin_make
Base path: /home/staff/peng/spot_ws
Source space: /home/staff/peng/spot_ws/src
Build space: /home/staff/peng/spot_ws/build
Devel space: /home/staff/peng/spot_ws/devel
Install space: /home/staff/peng/spot_ws/install
####
#### Running command: "cmake /home/staff/peng/spot_ws/src -DCATKIN_DEVEL_PREFIX=/home/staff/peng/spot_ws/devel -DCMAKE_INSTALL_PREFIX=/home/staff/peng/spot_ws/install -G Unix Makefiles" in "/home/staff/peng/spot_ws/build"
####
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/staff/peng/spot_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/staff/peng/spot_ws/devel;/home/staff/peng/catkin_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/staff/peng/spot_ws/devel;/home/staff/peng/catkin_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/staff/peng/spot_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
WARNING: package "moveit_task_constructor_core" should not depend on metapackage "moveit_planners" but on its packages instead
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - spot_description
-- ~~  - spot_driver
-- ~~  - spot_msgs
-- ~~  - spot_cam
-- ~~  - spot_viz
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'spot_description'
-- ==> add_subdirectory(spot_ros/spot_description)
-- +++ processing catkin package: 'spot_driver'
-- ==> add_subdirectory(spot_ros/spot_driver)
-- Installing devel-space wrapper /home/staff/peng/spot_ws/src/spot_ros/spot_driver/scripts/spot_ros to /home/staff/peng/spot_ws/devel/lib/spot_driver
-- Installing devel-space wrapper /home/staff/peng/spot_ws/src/spot_ros/spot_driver/test/ros_helpers_test.py to /home/staff/peng/spot_ws/devel/lib/spot_driver
-- Installing devel-space wrapper /home/staff/peng/spot_ws/src/spot_ros/spot_driver/test/spot_ros_test.py to /home/staff/peng/spot_ws/devel/lib/spot_driver
-- +++ processing catkin package: 'spot_msgs'
-- ==> add_subdirectory(spot_ros/spot_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[... 528 lines cut ...]
[ 37%] Generating Python from MSG spot_msgs/ImageProperties
[ 37%] Generating C++ code from spot_msgs/NavigateToFeedback.msg
[ 37%] Generating EusLisp code from spot_msgs/NavigateToResult.msg
[ 37%] Generating Lisp code from spot_msgs/NavigateToFeedback.msg
[ 37%] Generating Lisp code from spot_msgs/NavigateToResult.msg
[ 37%] Generating C++ code from spot_msgs/NavigateRouteAction.msg
[ 39%] Generating Javascript code from spot_msgs/NavigateToActionFeedback.msg
[ 39%] Generating Lisp code from spot_msgs/NavigateRouteAction.msg
[ 39%] Generating C++ code from spot_msgs/NavigateRouteActionGoal.msg
/home/staff/peng/spot_ws/src/spot_ros/spot_viz/src/look_at_point_tool.cpp:11:10: **fatal error: spot_cam/LookAtPoint.h: No such file or directory
   11 | #include <spot_cam/LookAtPoint.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [spot_ros/spot_viz/CMakeFiles/spot_viz.dir/build.make:89: spot_ros/spot_viz/CMakeFiles/spot_viz.dir/src/look_at_point_tool.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:6522: spot_ros/spot_viz/CMakeFiles/spot_viz.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....**

`

**Question 2: When I use conda virtual environment, there is the PyKDL issue still. I have done 'sudo apt install python3-pykdl' and it is already installed. I can find it at** 

`/usr/lib/python3/dist-packages$ ls | grep KDL
PyKDL.cpython-38-x86_64-linux-gnu.so`

Could you help to guide how to resolve this issue. By the way, I followed the instructions to built it locally, but failed with some missing header files.

heuristicus, 2023-11-09:
I've only tested the package with catkin build. Is there any particular reason why you need to use catkin make?

grapesonwang, 2023-11-09:
Thank you for your response.

 I am used to catkin_make so use it by default. Another reason is that because catkin_make had these issues, I was worried if catkin build has bypassed anything that I am not aware of, but which will cause issues in the future. If not, then I will use catkin build in the future.

Could you please help with the second question as well. Many thanks!

Question 2: When I use conda virtual environment, there is the PyKDL issue still. I have done 'sudo apt install python3-pykdl' and it is already installed. I can find it at

/usr/lib/python3/dist-packages$ ls | grep KDL PyKDL.cpython-38-x86_64-linux-gnu.so

Could you help to guide how to resolve this issue. By the way, I followed the instructions to built it locally, but failed with some missing header files.

grapesonwang, 2023-11-09:
> I've only tested the package with catkin build. Is there any particular reason why you need to use catkin make?

Thank you for your response.

 I am used to catkin_make so use it by default. Another reason is that because catkin_make had these issues, I was worried if catkin build has bypassed anything that I am not aware of, but which will cause issues in the future. If not, then I will use catkin build in the future.

Could you please help with the second question as well. Many thanks!

Question 2: When I use conda virtual environment, there is the PyKDL issue still. I have done 'sudo apt install python3-pykdl' and it is already installed. I can find it at

/usr/lib/python3/dist-packages$ ls | grep KDL PyKDL.cpython-38-x86_64-linux-gnu.so

Could you help to guide how to resolve this issue. By the way, I followed the instructions to built it locally, but failed with some missing header files.

heuristicus, 2023-11-09:
I'm not sure how this package will interact with a conda environment, and I'm not very familiar with how conda works in practice. The PyKDL issue has  been reported before as happening on melodic, but not on noetic. If you use the package outside a conda environment does it work? If so it may indicate that there's some setup issues with the environment.

grapesonwang, 2023-11-09:
> I'm not sure how this package will interact with a conda environment, and I'm not very familiar with how conda works in practice. The PyKDL issue has been reported before as happening on melodic, [...]

I have not tested outside conda environment yet but will do so next week and let you know by then! I had a look at PyKDL issue with melodic, probably different to mine but I will try to figure out! Many thanks!