heuristicus/spot_ros #150: Broken instructions/files

ryan-roche, 2024-06-11:
After following the instructions and building the `spot_driver` and `spot_viz` packages using `catkin build`, `roslaunch spot_driver driver.launch` errors out saying
Resource not found: spot_description
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/root/ros_ws/src/spot_ros/spot_driver
ROS path [2]=/root/ros_ws/src/spot_ros/spot_msgs
ROS path [3]=/root/ros_ws/src/spot_ros/spot_cam
ROS path [4]=/root/ros_ws/src/spot_ros/spot_viz
ROS path [5]=/opt/ros/noetic/share
The traceback for the exception was written to the log file

heuristicus, 2024-06-12:
Did you also build `spot_description`?

ryan-roche, 2024-06-13:
Manually building the `spot_driver`, `spot_viz`, and `spot_description` packages using `catkin build` works for the build step, but another error appears when attempting to launch the driver.

```
root@57cbf6900f0d:~/test_ws# roslaunch spot_driver driver.launch
... logging to /root/.ros/log/dcd21528-29bc-11ef-8437-0242ac110002/roslaunch-57cbf6900f0d-6385.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://57cbf6900f0d:38675/

SUMMARY
========

PARAMETERS
 * /bluetooth_teleop/joy_node/autorepeat_rate: 20
 * /bluetooth_teleop/joy_node/deadzone: 0.1
 * /bluetooth_teleop/joy_node/dev: /dev/input/js0
 * /bluetooth_teleop/teleop_twist_joy/axis_angular/yaw: 3
 * /bluetooth_teleop/teleop_twist_joy/axis_linear/x: 1
 * /bluetooth_teleop/teleop_twist_joy/axis_linear/y: 0
 * /bluetooth_teleop/teleop_twist_joy/enable_button: 4
 * /bluetooth_teleop/teleop_twist_joy/enable_turbo_button: 5
 * /bluetooth_teleop/teleop_twist_joy/scale_angular/yaw: 1.4
 * /bluetooth_teleop/teleop_twist_joy/scale_angular_turbo/yaw: 1.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear/x: 0.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear/y: 0.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear_turbo/x: 2.0
 * /bluetooth_teleop/teleop_twist_joy/scale_linear_turbo/y: 2.0
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /spot/spot_ros/allow_motion: True
 * /spot/spot_ros/auto_claim: False
 * /spot/spot_ros/auto_power_on: False
 * /spot/spot_ros/auto_stand: False
 * /spot/spot_ros/autonomy_enabled: True
 * /spot/spot_ros/claim/get_lease_on_action: False
 * /spot/spot_ros/claim/use_take_lease: False
 * /spot/spot_ros/depth_in_visual: True
 * /spot/spot_ros/estop_timeout: 9.0
 * /spot/spot_ros/hostname: 192.168.50.3
 * /spot/spot_ros/max_angular_velocity_z: 0.5
 * /spot/spot_ros/max_linear_velocity_x: 0.5
 * /spot/spot_ros/max_linear_velocity_y: 0.5
 * /spot/spot_ros/mode_parent_odom_tf: vision
 * /spot/spot_ros/password: dummypassword
 * /spot/spot_ros/rates/camera_images: 10.0
 * /spot/spot_ros/rates/check_subscribers: 10.0
 * /spot/spot_ros/rates/feedback: 10.0
 * /spot/spot_ros/rates/hand_image: 10.0
 * /spot/spot_ros/rates/lease: 1.0
 * /spot/spot_ros/rates/loop_frequency: 50.0
 * /spot/spot_ros/rates/metrics: 0.04
 * /spot/spot_ros/rates/mobility_params: 20.0
 * /spot/spot_ros/rates/point_cloud: 10.0
 * /spot/spot_ros/rates/robot_state: 20.0
 * /spot/spot_ros/rates/world_objects: 10.0
 * /spot/spot_ros/username: dummyusername
 * /twist_marker_server/linear_scale/x: 1
 * /twist_marker_server/linear_scale/y: 1
 * /twist_marker_server/link_name: body
 * /twist_marker_server/max_negative_linear_velocity/x: -1
 * /twist_marker_server/max_negative_linear_velocity/y: -1
 * /twist_marker_server/max_positive_linear_velocity/x: 1
 * /twist_marker_server/max_positive_linear_velocity/y: 1
 * /twist_marker_server/robot_name: spot
 * /twist_mux/locks: []
 * /twist_mux/topics: [{'name': 'bt_joy...

[... 42 lines cut ...]
ROS path [6]=/opt/ros/noetic/share
process[spot/spot_ros-6]: started with pid [6413]
ERROR: cannot launch node of type [twist_mux/twist_mux]: twist_mux
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/root/test_ws/src/spot_ros/spot_description
ROS path [2]=/root/test_ws/src/spot_ros/spot_driver
ROS path [3]=/root/test_ws/src/spot_ros/spot_msgs
ROS path [4]=/root/test_ws/src/spot_ros/spot_cam
ROS path [5]=/root/test_ws/src/spot_ros/spot_viz
ROS path [6]=/opt/ros/noetic/share
Traceback (most recent call last):
  File "/root/test_ws/devel/lib/spot_driver/spot_ros", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/root/test_ws/src/spot_ros/spot_driver/scripts/spot_ros", line 3, in <module>
    from spot_driver.spot_ros import SpotROS
  File "/root/test_ws/src/spot_ros/spot_driver/src/spot_driver/spot_ros.py", line 83, in <module>
    from spot_driver.ros_helpers import *
  File "/root/test_ws/src/spot_ros/spot_driver/src/spot_driver/ros_helpers.py", line 6, in <module>
    import transforms3d
ModuleNotFoundError: No module named 'transforms3d'
[spot/spot_ros-6] process has died [pid 6413, exit code 1, cmd /root/test_ws/devel/lib/spot_driver/spot_ros joint_states:=/joint_states tf:=/tf __name:=spot_ros __log:=/root/.ros/log/dcd21528-29bc-11ef-8437-0242ac110002/spot-spot_ros-6.log].
log file: /root/.ros/log/dcd21528-29bc-11ef-8437-0242ac110002/spot-spot_ros-6*.log
```

This was done in a fresh docker container using the official `osrf/ros:noetic-desktop-full` image. I manually installed `curl`, `wget` and catkin in order to build the packages.